pi-hole/PADD #279: "Slim" does not use all available rows.

yubiuser, 2022-12-05:
"Slim" is used when the console height is 21 (rows). However, we only use 20 and 1 of those is even empty.
https://github.com/pi-hole/PADD/blob/9218e0d587bdd8c3c3ab7ae2aeb75f7a9148af72/padd.sh#L734

We could use this space e.g. to display version information.
See comparison to regular, which needs 22 rows.

[image]
[image]

rdwebdesign, 2022-12-05:
Actually, Mega has the same problem.
80 columns, but only 79 are used.

rdwebdesign, 2022-12-05:
[image]

yubiuser, 2023-01-05:
Fixed with https://github.com/pi-hole/PADD/commit/ed3657f6a7c52e735c438cb534b13a9f73c9ea37 and https://github.com/pi-hole/PADD/pull/327

yubiuser, 2023-01-20:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.1

closed 2023-01-20 by yubiuser