pi-hole/PADD #287: Bug #273 is back?

Gothico74, 2022-12-12:
After updating PADD to 3.9 the headline shows Pi-Hole as aged, newer versions available.

pi@raspberrypi-hole:~ $ pihole -v
  Pi-hole version is v5.14.2 (Latest: v5.14.2)
  AdminLTE version is v5.17 (Latest: v5.17)
  FTL version is v5.19.2 (Latest: v5.19.2)

pi@raspberrypi-hole:~ $ sudo pihole -up
  [✓] Update local cache of available packages
  [i] Existing PHP installation detected : PHP version 7.3.31-1~deb10u1
  [✓] Checking for git
  [✓] Checking for iproute2
  [✓] Checking for dialog
  [✓] Checking for ca-certificates

  [i] Checking for updates...
  [i] Pi-hole Core:     up to date
  [i] Web Interface:    up to date
  [i] FTL:              up to date

  [✓] Everything is up to date!
[image]

rdwebdesign, 2022-12-12:
Can you check if  `pihole updatechecker` fix the issue?

Gothico74, 2022-12-12:
Yes i did, but no avail.
tryed everything that was mentioned in the old thread

rdwebdesign, 2022-12-12:
Ohhh... I think I found the issue:

```
- Gathering version information...
+ GetVersionInformation
++ echo v5.14.2
++ grep -oE '^v[0-9]+([.][0-9]+){1,2}'
+ CORE_VERSION=v5.14.2
+ '[' master = master ']'
+ '[' 00345387 = 0034538 ']'
+ out_of_date_flag=true
```

The comparison is failing with the same hash because there is a difference in the hash length.
```
00345387  = 8 characters
0034538   = 7 characters
```
This is happening with all 3 components.

yubiuser, 2022-12-12:
With `5.14.2` we ensure, that both local and remote hash should have a length of 8.

https://github.com/pi-hole/pi-hole/blob/0034538794a3754ab532a53480da39d3fd13d7ed/advanced/Scripts/updatecheck.sh#L23-L34

___

In any case, this is not a PADD issue but looks like Pi-hole one. I'll transfer the issue.