pi-hole/PADD #288: stty: standard input: Inappropriate ioctl for device

klutchell, 2022-12-12:
**Describe the bug**

Switching from `tput` to `stty size` results in errors if tty does not exist.

**To Reproduce**
Steps to reproduce the behavior:

1. Call PADD script from an s6 overlay in a docker container (eg. `/usr/src/app/padd.sh > /dev/tty1`)
2. ...

**Expected behavior**

The old method of calculating terminal size without a tty worked with no issues.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

- Introduced by: https://github.com/pi-hole/PADD/pull/268
- Similar issue fixed in: https://github.com/pi-hole/pi-hole/issues/145
- Downstream project issue: https://github.com/klutchell/balena-pihole/issues/187

rdwebdesign, 2022-12-12:
I was able to run PADD inside official Pi-hole containers.

Which image are you using?

klutchell, 2022-12-12:
Also the official Pi-hole container, with an external PiTFT display. Are you running the container as privileged?

Here's the compose file: https://github.com/klutchell/balena-pihole/blob/main/docker-compose.yml

rdwebdesign, 2022-12-12:
No. I'm running using the basic setup, but I didn't tested on a external display.
I only tested on different terminals.

klutchell, 2022-12-12:
Replacing
```
    console_height=$(stty size | awk '{ print $1 }')
    console_width=$(stty size | awk '{ print $2 }')
```
with
```
    console_height=$(tput lines)
    console_width=$(tput cols)
```
Which is what was working before seems to solve it on my devices. Do we know if the switch to `stty size` was intentional?

rdwebdesign, 2022-12-12:
> Do we know if the switch to stty size was intentional?

I think it was, but it needs confirmation.

yubiuser, 2022-12-12:
Yes, it was intentional.

Because of the (claimed) tput overhead:
https://github.com/dylanaraps/writing-a-tui-in-bash#escape-sequences

https://github.com/dylanaraps/writing-a-tui-in-bash#using-stty

rdwebdesign, 2022-12-12:
Did you saw this?
https://github.com/docker/compose/issues/1876

`tty:true` should fix it.

klutchell, 2022-12-13:
`tty: true` is set by default by the balena supervisor (I confirmed by inspecting the container).

However I have a [workaround](https://github.com/klutchell/balena-pihole/pull/188) by using `stty size -F /dev/tty1` so we can close this if no one else is experiencing any issues.

yubiuser, 2022-12-13:
Let's keep it open until it goes auto-stale in 30 days. Just to make sure users can find it easier if they experience the same issue.

yubiuser, 2022-12-29:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.0

closed 2022-12-29 by yubiuser