pi-hole/PADD #290: Centred layot makes screen flicker

FirstGeekDanny, 2022-12-12:
First off let me just day, I LOVE the centred layout.

I have the official 7" Pi TFT screen and while the centred layout is gorgrous, the display now flickers... a LOT!

FirstGeekDanny, 2022-12-12:
Okay, rebooted the pi with the changed config and it still flickers.

rdwebdesign, 2022-12-13:
Thanks for the test.

We will try to find out what is causing the flickering.

yubiuser, 2022-12-13:
Please change the following lines locally

https://github.com/pi-hole/PADD/blob/89853fae684e65f2fc02177fd7ef06fbbaaf5019/padd.sh#L935-L936

to

```
    console_width=$(tput cols)
    console_height=$(tput lines)
```

and see if this fixes the issue. You can (additionally) try to remove

https://github.com/pi-hole/PADD/blob/89853fae684e65f2fc02177fd7ef06fbbaaf5019/padd.sh#L934

FirstGeekDanny, 2022-12-13:
Hmm, I'm a windows geek, not very good with editing things on linux.

yubiuser, 2022-12-13:
Use the `nano` editor. `Ctrl + o` saves the file, `Ctrl + x` closes the editor.

FirstGeekDanny, 2022-12-13:
Flickers a lot less, but still flickering...

yubiuser, 2022-12-13:
Which of the two recommendation did you implement?

yubiuser, 2022-12-13:
Could you make a video/screenrecording of the PADD flickering? So we could see when and how often the flickering is happening.

FirstGeekDanny, 2022-12-13:
I implemented both of them...

[https://1drv.ms/u/s!AvfBkai12iha9OsEObM2FiRxmomoRw?e=QZQXmP](url)

yubiuser, 2022-12-14:
It's basically flickering once every 5 seconds before the dashboard is updated.

LPanic, 2022-12-15:
I can confirm, having the same issue. Tried the proposed fix. Flickering improves, but not gone. I can support in testing.

accordxtc, 2022-12-15:
I too am having the issue with a Elecrow 5" screen

DL6ER, 2022-12-15:
@accordxtc Is it [this](https://www.elecrow.com/hdmi-5-inch-800x480-tft-display-for-raspberry-pi-b-p-1384.html) one? If so, I'll try to buy one so we can better try to reproduce the issue ourselves.

accordxtc, 2022-12-15:
@DL6ER I have the Elecrow 5" IPS touch screen model DIS00005R

PromoFaux, 2022-12-15:
https://www.amazon.co.uk/Elecrow-Capacitive-Interface-Supports-Raspberry/dp/B07PQQGP7R ?

DL6ER, 2022-12-15:
@accordxtc 

> Elecrow 5" IPS touch screen model DIS00005R

I ordered one of them and we'll use it to verify our fix works as expected.

accordxtc, 2022-12-15:
@DL6ER Thanks, before 3.9.0 it was working with no flicker

yubiuser, 2022-12-15:
Please try if this special version fixes the issue

https://github.com/pi-hole/PADD/blob/no_flicker/padd.sh

accordxtc, 2022-12-15:
@yubiuser Nope this wont even launch padd and gives line 0: a syntax error near unexpected token 'newline'

rdwebdesign, 2022-12-15:
Did you replaced the original `padd.sh` with the new one?

accordxtc, 2022-12-15:
@rdwebdesign Yes

rdwebdesign, 2022-12-15:
Just to be sure, the new code is on this link:
**https://raw.githubusercontent.com/pi-hole/PADD/no_flicker/padd.sh**

The previous link posted above is linking to the gihub page to show the new code, but it's an HTML page, not the real code.

accordxtc, 2022-12-15:
@rdwebdesign Using your link worked. Screen flicker is gone for me

DL6ER, 2022-12-16:
I received the display about half an hour ago. My observations are:

1. Using a terminal in in the graphical interface: It does not flicker on `master` regardless of being in a window or being full-screen

[image]
 
3. Using the virtual console (<kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>F1</kbd>): It ***does*** flicker on `master`
2. Using the virtual console (<kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>F1</kbd>): It ***does not*** flicker on `no_flicker`

https://user-images.githubusercontent.com/16748619/208118966-6c73968b-d12a-4cc8-87eb-6eb1357536e6.mp4

@yubiuser I'd be ready for approval when you open a PR.

FirstGeekDanny, 2022-12-16:
Yup, it's fixed on my end too... no flickering here anymore.

FirstGeekDanny, 2022-12-16:
If you leave it for a little while the layout goes a little crazy... but it still doesn't flicker...
[image]

yubiuser, 2022-12-16:
Thanks for the report. This is a different issue, just revealing itself now as we don't clear the whole screen anymore in the fixing branch.

yubiuser, 2022-12-16:
Mhh.. looks like `ps XXX` failed. However, we don't use this command. Are you running something else on your Pi?

FirstGeekDanny, 2022-12-16:
It's just the pihole with Cloudflared

yubiuser, 2022-12-17:
I was wrong, we use `ps`

https://github.com/pi-hole/PADD/blob/6676f562e074336241bd06807c60718bafa1b987/padd.sh#L416-L417

kagegoomba, 2022-12-18:
I'm seeing a flicker on mine and the "no_flicker" links are reporting 404 errors when I try to pull the file to test.

yubiuser, 2022-12-18:
`no_flicker` was deleted after it got merged into development. We'll release a bugfix release soon.

yubiuser, 2022-12-29:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.0

closed 2022-12-29 by yubiuser