pi-hole/PADD #293: Issue with product_family in docker on a Synology nas

PromoFaux, 2022-12-15:
The below line path does not appear to exist when I run on docker on my Synology NAS, however I do not see the same thing in docker on a raspberry Pi

https://github.com/pi-hole/PADD/blob/89853fae684e65f2fc02177fd7ef06fbbaaf5019/padd.sh#L222

[image]

rdwebdesign, 2022-12-15:
How is this line executed?

Line 219 should be `false`:

https://github.com/pi-hole/PADD/blob/89853fae684e65f2fc02177fd7ef06fbbaaf5019/padd.sh#L219

rdwebdesign, 2022-12-15:
Actually it should be:
```
-  if [ -f /sys/devices/virtual/dmi/id/product_name ] || [ -f /sys/devices/virtual/dmi/id/product_family ]; then 
+  if [ -f /sys/devices/virtual/dmi/id/product_name ] && [ -f /sys/devices/virtual/dmi/id/product_family ]; then 
```

PromoFaux, 2022-12-15:
Because it's hitting the `||` condition.

`product_name` is populated:

```
admin@fappotron:~$ docker exec -it pihole cat /sys/devices/virtual/dmi/id/product_name
DS218+
admin@fappotron:~$ docker exec -it pihole cat /sys/devices/virtual/dmi/id/product_family
cat: /sys/devices/virtual/dmi/id/product_family: No such file or directory
```

rdwebdesign, 2022-12-15:
So, we need a different test, to avoid the use of an absent file:

```sh
  if [ -f /sys/devices/virtual/dmi/id/product_name ]; then
    # Get model, remove possible null byte
    product_name=$(tr -d '\0' < /sys/devices/virtual/dmi/id/product_name)
    
    sys_model="$(echo "$product_name")"
    if [ -f /sys/devices/virtual/dmi/id/product_family ]; then
      product_family=$(tr -d '\0' < /sys/devices/virtual/dmi/id/product_family)
      sys_model="$(echo "$product_name" | grep "$product_family")"
    fi

    # If product_family is not contained in product_name, both are shown
    if [ -z "$sys_model" ]; then
      sys_model="${product_family} ${product_name}"
    fi
  elif [ -f /sys/firmware/devicetree/base/model ]; then
```

rdwebdesign, 2022-12-15:
@yubiuser 

Promofaux found a case where `product_name` exists, but `product_family` does not exist.

In this case, the current code fails.
I suggested a new code that works if both files exist or only `product_name` exists.

Are there cases where only `product_family` exists?
If the answer is positive (or it's not possible to predict), we need to change the code to guarantee we read the files only if they exist.

yubiuser, 2022-12-15:
> Are there cases where only product_family exists?

Actually, I don't know - and worse: I don't know how to test this. We can't check all devices that exist out there. Therefore we should assume there are devices that have only a `product_family` set.

PromoFaux, 2022-12-15:
Do we even need it? It's only used in `mega` as far as I can see

https://github.com/pi-hole/PADD/blob/6676f562e074336241bd06807c60718bafa1b987/padd.sh#L857

rdwebdesign, 2022-12-15:
Because mega is bigger and we can show more information, like this.

PromoFaux, 2022-12-15:
It just seems like a lot of code to possibly spit out `Unknown` in cases where the screen happens to be big enough

https://github.com/pi-hole/PADD/blob/6676f562e074336241bd06807c60718bafa1b987/padd.sh#L218-L268

Screenshot from a raspberry pi 4

[image]

rdwebdesign, 2022-12-15:
This is the PR, including the discussion:
https://github.com/pi-hole/PADD/pull/267

PromoFaux, 2022-12-15:
Hmmm, the code that was removed would work on an Rpi:

```
pi@dockerpi:~ $ cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.4
```

So was the PR to change it so that device name could _only_ be derived on an x86 machine?

yubiuser, 2022-12-15:
No.

The check is still in there.
https://github.com/pi-hole/PADD/blob/6676f562e074336241bd06807c60718bafa1b987/padd.sh#L229

PromoFaux, 2022-12-15:
My bad - I missed that entirely! OK, if I extract that I can see the model number when running directly on the pi:

```
pi@dockerpi:~ $ ./test.sh
Raspberry Pi 4 Model B Rev 1.4
```

But from within the container it does not work

```
pi@dockerpi:~ $ docker exec -it pihole cat /sys/firmware/devicetree/base/model
cat: /sys/firmware/devicetree/base/model: No such file or directory
```

No biggy I guess!

rdwebdesign, 2022-12-15:
Inside containers are very difficult to identify the hardware.

yubiuser, 2022-12-15:
Figuring out the base device from within a script running in a docker contains might be out-of-scope for `PADD`

rdwebdesign, 2022-12-15:
I think it's possible (or at least easier) if the docker socket is mounted inside the container, but this is completely out of question, because it is very unsafe.

It could be possible using external software, but I think we don't want to add a new dependency just to be able to get this information.

PromoFaux, 2022-12-15:
I already think it's a lot of code just for outputting the device - imagine what I would think about adding in special cases for docker ;)

yubiuser, 2022-12-15:
Most is the code for cleaning unwanted stuff.

rdwebdesign, 2022-12-16:
**Note:**
tested only on Raspberry Pi.

The command `cat /proc/cpuinfo | grep "Model"` returns <code>Model&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;: Raspberry Pi 4 Model B Rev 1.4</code>, even inside a container.

yubiuser, 2022-12-17:
`cat /proc/cpuinfo | grep "Model"` does not return anything on my `x86_64` laptop nor my RockPi 4b. 

Do you think we should add it as an additional source?

rdwebdesign, 2022-12-17:
I don't think this will work... Inside a container is not possible to access all hardware info (this is intentional, for security reasons).

I added the code to show the Docker Tag inside containers.

DL6ER, 2022-12-18:
@yubiuser Just seen this and thought it'd be worthwhile to point out that the key is lowercase in the kernel (but you `grep`ped for uppercase). Try `grep -i "model" < /proc/cpuinfo` probably even `grep -i "model name" < /proc/cpuinfo`

yubiuser, 2022-12-18:
Better, but not good ;-)
```
rockpi@rockpi-4b:~$ grep -i "model" < /proc/cpuinfo
rockpi@rockpi-4b:~$ grep -i "model name" < /proc/cpuinfo
```

```
chrko@ThinkPad-X230:~$ grep -i "model" < /proc/cpuinfo
model		: 58
model name	: Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz
model		: 58
model name	: Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz
model		: 58
model name	: Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz
model		: 58
model name	: Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz

```

DL6ER, 2022-12-19:
So ... this raises the question: What is the content of `/proc/cpuinfo` on `rockpi-4b` at all?

DL6ER, 2022-12-19:
Some further ideas:

### dmidecode --type processor
```
sudo dmidecode --type processor | sed -n "s/Version: \(.*\)/\1/p"
AMD Ryzen 7 4800H with Radeon Graphics
```

### lshw -C CPU
```
$ sudo lshw -C CPU 2> /dev/null | sed -n "s/product: \(.*\)/\1/p"
AMD Ryzen 7 4800H with Radeon Graphics
```

### lscpu
```
$ lscpu | sed -n "s/Model name:\W*\(.*\)/\1/p"
AMD Ryzen 7 4800H with Radeon Graphics
```

If they return empty for you, you should check what the commands return without `sed`

rdwebdesign, 2022-12-19:
Are out sure this commands are available everywhere?

None of the commands above are available inside our container.

Since we are going to add PADD on `docker-pi-hole` images, to I think we need to separate these tests:
- if inside a docker container: show image tag;
- else, use those other methods.

yubiuser, 2022-12-19:
> What is the content of /proc/cpuinfo

```
rockpi@rockpi-4b:~$ cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 4
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 2

processor	: 5
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 2
```
____
```
rockpi@rockpi-4b:~$ sudo dmidecode --type processor | sed -n "s/Version: \(.*\)/\1/p"
rockpi@rockpi-4b:~$ sudo lshw -C CPU 2> /dev/null | sed -n "s/product: \(.*\)/\1/p"
rockpi@rockpi-4b:~$ lscpu | sed -n "s/Model name:\W*\(.*\)/\1/p"
Cortex-A53
```

rdwebdesign, 2022-12-19:
`dmidecode` doesn't work on my 2 Raspberry Pi (running on Buster):

```console
pi@pi4:~ $ sudo dmidecode
# dmidecode 3.2
Scanning /dev/mem for entry point.
# No SMBIOS nor DMI entry point found, sorry.
```

`lshw` is not installed by default.

`lscpu | sed -n "s/Model name:\W*\(.*\)/\1/p"` returns the same CPU model: `Cortex-A53`

yubiuser, 2022-12-29:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.0

closed 2022-12-29 by yubiuser