pi-hole/PADD #303: `ps` failed

yubiuser, 2022-12-17:
If you leave it for a little while the layout goes a little crazy... but it still doesn't flicker...
[image]

_Originally posted by @FirstGeekDanny in https://github.com/pi-hole/PADD/issues/290#issuecomment-1355096366_

yubiuser, 2022-12-17:
Somwhow `ps` failed.

https://github.com/pi-hole/PADD/blob/6676f562e074336241bd06807c60718bafa1b987/padd.sh#L416-L417

rdwebdesign, 2022-12-18:
If `${ftlPID}` is empty, I see the same error, but I don't know how this is happening.

rdwebdesign, 2022-12-18:
As a workaround we can add `2>/dev/null` to avoid the error:
```diff
  ftl_cpu="$(ps -p "${ftlPID}" -o %cpu 2>/dev/null | tail -n1 | tr -d '[:space:]')" 
  ftl_mem_percentage="$(ps -p "${ftlPID}" -o %mem 2>/dev/null | tail -n1 | tr -d '[:space:]')" 
```

rdwebdesign, 2022-12-18:
I think the issue happens if FTL is restarted exactly at the same time PADD updates the screen.
The new PID is not updated immediately and the `ps` command fails showing an error message. After that, the error never goes away until restart PADD.

I saw a similar issue on PHP, when FTL is restarted via web interface.

rdwebdesign, 2022-12-18:
We can also change the function "moveXOffset()" to print N spaces, instead of moving the cursor:

```diff sh
- printf '\e[%sC' "${xOffset}"
+ printf '%*s' "${xOffset}"
```

DL6ER, 2022-12-18:
The test is broken. We should fix it rather than adding a workaround in form of `2> /dev/null`

``` bash
if [ -z ${ftlPID+x} ]; then
```
will *always* succeed as `+x` will substitute the variable by `x` if the variable is set. As we have
```
ftlPID=$(pidof pihole-FTL)
```
the variable will never be unset, but if it empty (FTL not running), the substitution will kick in and the test will succeed.

The test should read
``` bash
if [ -z "${ftlPID}" ]; then
```
instead.

yubiuser, 2022-12-29:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.0

PR #304 (2022-12-18): Fix check if FTL is running
**What does this PR aim to accomplish?:**

Fix #303 

**How does this PR accomplish the above?:**

The currently used test is broken:

``` bash
if [ -z ${ftlPID+x} ]; then
```
will *always* succeed as `+x` will substitute the variable by `x` if the variable is set.

As we have
```
ftlPID=$(pidof pihole-FTL)
```
above, the variable will never be unset. However, when FTL ist not running and the variable is empty, the substitution will kick in and the test will still succeed (although it obviously shouldn't). This test is simply wrong.

The test should read
``` bash
if [... cut]
closed 2022-12-29 by yubiuser