pi-hole/PADD #305: Versions are showing Red Once Again

kagegoomba, 2022-12-18:
Just like last time - after updating to the latest version I'm seeing "Red Version Numbers" on "Current" Pi-Hole versions - I cat'ed the versions file and its all matching current builds across the board - also did a pi-hole update-checker as well.

Course I mentioned in the other branch that my screen is now flickering every 5 or so seconds which it seems you have a fix in the pipe - coincidentally I see they also have the same red version numbers too. :)

[image]

yubiuser, 2022-12-18:
Fixed with https://github.com/pi-hole/PADD/pull/302

kagegoomba, 2022-12-20:
Is there an ETA on when the next release is coming? Would like to stop twitching at my PADD flickering. :)

yubiuser, 2022-12-20:
I'd  like to get https://github.com/pi-hole/PADD/pull/296 and https://github.com/pi-hole/PADD/pull/306 in. But both need review and time is limited before Christmas. You can already use

https://raw.githubusercontent.com/pi-hole/PADD/development/padd.sh

kagegoomba, 2022-12-20:
I'll give it a go later today. Much Thanks.

kagegoomba, 2022-12-21:
Just tried it - so much better.
Now I don't feel like I'm losing my mind or having a seizure.

Also the version numbers are green.

Much thanks!

yubiuser, 2022-12-29:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.0

closed 2022-12-29 by yubiuser