pi-hole/PADD #307: `DNS is off!` rather than `System is healthy`.

yubiuser, 2022-12-19:
@yubiuser When using `development` branch of `PADD` (as of yesterday) I'm getting a new bug where I see `DNS is off!` rather than `System is healthy`. 

This changed a few hours after switching (things were fine initially, but changed over night).

There's no actual issue with DNS, the PiHole is working as expected and is blocking normally.

Edit 1: I just rebooted and see `System is healthy` again. Will report back if I see it change again.

Edit 2: `DNS is off!` is back 15 minutes after reboot.

[image]

_Originally posted by @christianboyle in https://github.com/pi-hole/pi-hole/issues/5059#issuecomment-1357768156_

yubiuser, 2022-12-19:
This is interesting. We fixed a bug in `PADD` that would always report `FTL` as running in https://github.com/pi-hole/PADD/pull/304

yubiuser, 2022-12-19:
The only condition that can lead to `DNS is off` while `FTL: Running` is when FTL is running but `"dns-port"` returned from the API is `0`(meaning DNS is switched off)

https://github.com/pi-hole/PADD/blob/1cf30ee904fb7501eb1ac98f56bd833a25496275/padd.sh#L425-L426

christianboyle, 2022-12-19:
Very strange. This is the first time I've seen this message and haven't made any changes to my network recently. I guess we'll just have to see if any others run into this with the new version.

christianboyle, 2022-12-22:
[image]

Inexplicably, this issue resolved itself after being broken for 2 days. Not sure if it was something intermittent, but will keep an eye out in case it happens again. 

I assume we can close this issue for now.

rdwebdesign, 2022-12-22:
If you need, we can be reopen the issue.

christianboyle, 2022-12-22:
[image]

Looks like things changed back over night. Not sure what's going on as I still haven't had any changes to my network or PiHole.

yubiuser, 2022-12-22:
Ok, I see a possible pitfall we created:

https://github.com/pi-hole/PADD/blob/1cf30ee904fb7501eb1ac98f56bd833a25496275/padd.sh#L423

askes for returning the current DNS port. The function will return `0` if DNS is disabled in `pihole-FTL`. However, we also return `0` in `GetFTLData()` if we fail to get the telnet API port

https://github.com/pi-hole/PADD/blob/1cf30ee904fb7501eb1ac98f56bd833a25496275/padd.sh#L114-L124

I don't know why we are failing to get the telnet port yet, but I'll prepare a fix to use `0` both as signal for failure and as actual data.

yubiuser, 2022-12-22:
What is the content of your `/etc/pihole/pihole-FTL.conf`?

christianboyle, 2022-12-22:
```
#; Pi-hole FTL config file
#; Comments should start with #; to avoid issues with PHP and bash reading this file
RATE_LIMIT=1000/60
PRIVACYLEVEL=0
```

yubiuser, 2022-12-22:
Ok, this looks not suspicious. Could you check your `/var/log/pihole/FTL.log` for any errors or (unintended restarts ) please?

christianboyle, 2022-12-22:
```
[2022-12-22 02:28:16.565 887M] Resizing "FTL-dns-cache" from 143360 to (9216 * 16) == 147456 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-22 08:15:33.662 887M] Resizing "FTL-dns-cache" from 147456 to (9472 * 16) == 151552 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-22 10:18:28.352 887M] Resizing "FTL-dns-cache" from 151552 to (9728 * 16) == 155648 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-22 11:14:42.058 887M] Resizing "FTL-dns-cache" from 155648 to (9984 * 16) == 159744 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-22 11:35:47.581 887M] Resizing "FTL-domains" from 143360 to (8192 * 20) == 163840 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-22 11:53:17.596 887M] WARNING in dnsmasq core: reducing DNS packet size for nameserver 2001:4860:4860::8888 to 1232
[2022-12-22 11:54:17.715 887M] Resizing "FTL-dns-cache" from 159744 to (10240 * 16) == 163840 (/dev/shm: 7.0MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-22 15:30:44.377 887M] Resizing "FTL-strings" from 245760 to (286720 * 1) == 286720 (/dev/shm: 7.0MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-22 15:44:04.615 887M] Resizing "FTL-dns-cache" from 163840 to (10496 * 16) == 167936 (/dev/shm: 7.0MB used, 483.5MB total, FTL uses 7.0MB)
```

rdwebdesign, 2022-12-22:
Look at `/var/log/pihole/FTL.log.1` to search for errors from yesterday.

christianboyle, 2022-12-22:
`FTL.log.1`

```
[2022-12-21 01:16:55.599 887M] Resizing "FTL-dns-cache" from 110592 to (7168 * 16) == 114688 (/dev/shm: 6.8MB used, 483.5MB total, FTL uses 6.8MB)
[2022-12-21 07:48:57.925 887M] Resizing "FTL-dns-cache" from 114688 to (7424 * 16) == 118784 (/dev/shm: 6.8MB used, 483.5MB total, FTL uses 6.8MB)
[2022-12-21 09:11:47.495 887M] Resizing "FTL-strings" from 204800 to (245760 * 1) == 245760 (/dev/shm: 6.8MB used, 483.5MB total, FTL uses 6.8MB)
[2022-12-21 09:28:35.993 887M] WARNING in dnsmasq core: reducing DNS packet size for nameserver 8.8.8.8 to 1232
[2022-12-21 10:54:35.014 887M] Resizing "FTL-dns-cache" from 118784 to (7680 * 16) == 122880 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-21 11:12:15.912 887M] Resizing "FTL-dns-cache" from 122880 to (7936 * 16) == 126976 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-21 11:28:48.320 887M] Resizing "FTL-domains" from 122880 to (7168 * 20) == 143360 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-21 13:09:15.353 887M] Resizing "FTL-dns-cache" from 126976 to (8192 * 16) == 131072 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-21 15:55:24.569 887M] Resizing "FTL-dns-cache" from 131072 to (8448 * 16) == 135168 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-21 19:44:01.224 887M] Resizing "FTL-dns-cache" from 135168 to (8704 * 16) == 139264 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
[2022-12-21 20:40:40.957 887M] Resizing "FTL-dns-cache" from 139264 to (8960 * 16) == 143360 (/dev/shm: 6.9MB used, 483.5MB total, FTL uses 6.9MB)
```

baz069, 2022-12-23:
+1 for seeing this issue. Currently running dev branch whilst waiting on the fix for #305  'versions showing red once again'.

christianboyle, 2022-12-23:
Last night I updated PiHole with `pihole -up` and then updated my Debian Bullseye packages `sudo apt update && sudo apt upgrade`.

I'm still using the `development` branch of `PADD`.

Everything is good so far this morning, seeing `System is healthy` and not `DNS is off!`.

I hadn't updated my Bullseye packages in a while (months), so maybe there was something out-of-date there that was impacting FTL, not sure.

Will keep an eye on it and report back if `DNS is off!` returns.

yubiuser, 2022-12-23:
Thanks for the log output - everything you see in there is normal and does not cause the issue.

christianboyle, 2022-12-23:
Well that's good, but `DNS is off!` just showed back up out of nowhere.

yubiuser, 2022-12-23:
I prepared a "debug" version that will help us to see which of the statements of

https://github.com/pi-hole/PADD/blob/1cf30ee904fb7501eb1ac98f56bd833a25496275/padd.sh#L425-L426

is true when the switch happens. All it does is to write `{ftlPID} and `${ftl_dns_port}` to `/tmp/PADD_debug.log`.
___

The version is at https://raw.githubusercontent.com/pi-hole/PADD/debug/padd.sh.

Please run this and let us now what it prints in the "normal" and "faulty" state. 
(Make sure you have enough space in `/tmp` as the function will print a line every 5 seconds)

christianboyle, 2022-12-23:
When in "normal" state (when `System is healthy`): 

`PID 857, DNS port 53` (repeating)

Will report back when it changes to `DNS is off!`

christianboyle, 2022-12-24:
Hmm, this is odd, but it we are seeing `DNS is off!`

`cat /tmp/PADD_debug.log`
`PID 857, DNS port 53`

yubiuser, 2022-12-24:
I think one issue and a bug (discovered by @rdwebdesign) come together here. Please check the whole debug file if there is at least **one** line that looks different. I suspect the following: at some point PADD fails in the communication with `FTL` setting `DNS is off`. In the next iteration everything is back to normal, however (and this is the bug) PADD fails to revert the message to "Everything is good". 
We already know how to solve the bug (bit it will need a bigger restructuring), but not why the communication failed.

yubiuser, 2022-12-24:
This should fix the status message not returning to healthy after an outage:
https://raw.githubusercontent.com/pi-hole/PADD/fix/status/padd.sh

christianboyle, 2022-12-24:
[image]

It looks like at some point the `PID` gets lost (all lines before and after this are the same).

yubiuser, 2022-12-24:
Thanks. This is the line I was looking for. Unfortunately I don't think we can do much about it. We use a very basic system command to get the `PID`. I don't see any reason why it should fail once and then continue working and returning the same `PID`.
https://github.com/pi-hole/PADD/blob/6676f562e074336241bd06807c60718bafa1b987/padd.sh#L401

Interestingly, a few lines down we are able to communicate with `FTL` to get
https://github.com/pi-hole/PADD/blob/6676f562e074336241bd06807c60718bafa1b987/padd.sh#L421

__

To sum it up: the issue is very likely outside of `PADD` but a additional `PADD` but lead to the behavior you see. This should be fixed by the version linked above.

yubiuser, 2022-12-24:
> Unfortunately I don't think we can do much about it. We use a very basic system command to get the PID. I don't see any reason why it should fail once and then continue working and returning the same PID.

I was wrong about this: we can change the command, as `pidof` seems to be unreliable sometimes (see https://unix.stackexchange.com/questions/518411/why-is-pidof-not-working and https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=926896).

__ 

After you hopefully confirmed the branch `fix/status` mitigated the issue, please switch back to https://raw.githubusercontent.com/pi-hole/PADD/debug/padd.sh

I added a commit using `ps` instead of `pidof`. Let's see if this works better.

christianboyle, 2022-12-24:
Nice, will switch back to `debug` version tonight or tomorrow morning (to give the bug adequate time to surface again).

christianboyle, 2022-12-25:
[image]

No change overnight, so I think we can confirm `fix/status` branch fixed the issue.

But after switching to `debug` branch I now get this error flashing every 5 seconds.

rdwebdesign, 2022-12-25:
Can you try to replace the `ps h` with `ps --no-headers` and test it again?

christianboyle, 2022-12-25:
Same thing with `--no-headers`.

I played around with this for a few minutes, but kept getting the `improper list` error when using `ps h -C pihole-FTL -o %p` in `padd.sh`.

I'm not sure if it matters, but it seems like there are 2 character spaces that appear in the output when running `ps h -C pihole-FTL -o %p` at the command line, maybe this is triggering the `improper list` issue, because it's reading those spaces as part of the `PID`?

[image]

As far as I can tell, `pidof` is the better option here (`fix/status` branch), as it doesn't have the flashing `improper list` issue.

yubiuser, 2022-12-25:
Interesting, I don't have those spaces on my installation. We could easily remove them, but I created a third branch, called `ps` that uses a completely different method to get the PID (same as we use for `Pi-hole` itself). If it will fail, you should get "DNS is off" stucked again.

https://raw.githubusercontent.com/pi-hole/PADD/ps/padd.sh

christianboyle, 2022-12-25:
Yea, trimming those spaces seems to have worked (no more `improper list` flashing).

[image]

christianboyle, 2022-12-25:
Things seem good so far on `ps` branch. `System is healthy` and no errors or flashing related to `improper list`. 

Not sure I'll notice if `DNS is off!` comes back, if this fix successfully updates the `System is healthy` message upon subsequent check (unless I happen to look at just the right time).

Should we add logging back to this branch to see if everything works as expected with this new fix? Or do we think `DNS is off!` is unlikely to show again, due to how the `PID` check is done now?

yubiuser, 2022-12-25:
> Not sure I'll notice if DNS is off! comes back, if this fix successfully updates the System is healthy message upon subsequent check (unless I happen to look at just the right time).

`ps` does not include the changes from `fix/status`. Therefore, if it fails once, you'll get the `DNS is off` forever again and you'll notice

christianboyle, 2022-12-26:
No issues after ~22 hours, I think it's safe to say `ps` branch is working as expected.

yubiuser, 2022-12-26:
Thanks for letting us know. We now have two ways (`ps` and `fix/status`) to solve this issue. Both can and should be part of PADD.

yubiuser, 2022-12-29:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.0

closed 2022-12-29 by yubiuser