pi-hole/PADD #318: PADD 3.9.0 reports Pi-hole 5.14.2 as out of date

BoscaBeag, 2022-12-29:
**Describe the bug**
After updating
```
AdminLTE version is v5.18 (Latest: v5.18) from 5.17.2
FTL version is v5.20 (Latest: v5.20) from 5.19.2
PADD version is 3.9.0 (Latest: v3.9.0) from 3.8.1
No upgrade of Pi-hole version is v5.14.2 (Latest: v5.14.2)
```

PADD reports 
`Pi-hole® v5.14.2, Web v5.18, FTL v5.20, PADD v3.9.0`
with the Pi-hole version colored red, and `[i] Updates are available`

PADD 3.8.1 correctly indicates Pi-hole as green, and PADD as out-of-date.

**To Reproduce**
Steps to reproduce the behavior:

1. pihole -up
2. padd_3.9.0.sh
3. padd_3.8.1.sh

**Expected behavior**
That the versions are correctly indicated as being up to date, or out of date.

**Additional context**

existing version of /etc/pihole/versions was
```
CORE_BRANCH=master
WEB_BRANCH=master
FTL_BRANCH=master
CORE_VERSION=v5.14.2
WEB_VERSION=v5.18
FTL_VERSION=v5.20
GITHUB_CORE_VERSION=v5.14.2
GITHUB_WEB_VERSION=v5.18
GITHUB_FTL_VERSION=v5.20
CORE_HASH=00345387
GITHUB_CORE_HASH=8d2e023e
WEB_HASH=ad2d4386
GITHUB_WEB_HASH=ad2d4386
FTL_HASH=cba74934
GITHUB_FTL_HASH=cba74934
```

after removing /etc/pihole/versions and running `pihole updatechecker`, the new versions file
```
CORE_VERSION=v5.14.2
CORE_BRANCH=master
CORE_HASH=00345387
GITHUB_CORE_VERSION=v5.14.2
GITHUB_CORE_HASH=8d2e023e
WEB_VERSION=v5.18
WEB_BRANCH=master
WEB_HASH=ad2d4386
GITHUB_WEB_VERSION=v5.18
GITHUB_WEB_HASH=ad2d4386
FTL_VERSION=v5.20
FTL_BRANCH=master
FTL_HASH=cba74934
GITHUB_FTL_VERSION=v5.20
GITHUB_FTL_HASH=cba74934
```

Please note in both that the local hash for CORE is **"00345387"** (same as @infinitytec in https://github.com/pi-hole/pi-hole/issues/4982) compared to the GITHUB_CORE_HASH of **"8d2e023e"**

rdwebdesign, 2022-12-29:
This is a known issue and it's already fixed on development (https://github.com/pi-hole/PADD/pull/302).

yubiuser, 2022-12-29:
Duplicate of https://github.com/pi-hole/PADD/issues/305

closed 2022-12-29 by yubiuser