pi-hole/PADD #324: "Updates are available" when using Pi-hole in Docker Container but "unable" to update PADD

thomasmerz, 2023-01-01:
**Describe the bug**
I'm running Pi-hole in a Docker Container with "latest" everything. But `PADD` could be updated, but it couldn't be updated until Docker Image will get this update. This might irritate or "frustate" users:
```
 __      __  __   
|__) /\ |  \|  \     Pi-hole® v5.14.2, Web v5.18, FTL v5.20, PADD v3.9.0
|   /--\|__/|__/     [✓] Core  [✓] FTL   [i] Updates are available
```

**Expected behavior**
I'm not exactly sure how `PADD` should inform Docker users that there is an update available for `PADD`… Maybe **let's discuss if this is not neccessary for Docker because Docker Images should always be considered as an All-in-One**!? 🤔  Or do you prefer an update-strategy **inside** a running Docker Container?

**Screenshots**
`docker exec -it pihole padd`
[image]

https://github.com/pi-hole/PADD/
[image]

https://github.com/pi-hole/docker-pi-hole/
[image]

PromoFaux, 2023-01-01:
Ah, I had not realised there was a new version tagged - busy time of year. 

New docker image building as we speak (We'll pretend that it was tagged 2 days ago 😉 )
https://github.com/pi-hole/docker-pi-hole/actions/runs/3817491391

I think there should be something added to PADD that says "If it is inside docker, don't display the update notice", like we have for the web/core components

PromoFaux, 2023-01-01:
> but it couldn't be updated until Docker Image will get this update.

Technically it could be overwritten inside the container if one was desperate for the latest PADD. It lives at `/usr/local/bin/padd` inside the container. Obviously any changes made would be lost on restarting the container.

thomasmerz, 2023-01-02:
That's not what containers are made for (inplace-patching)… Ideas so far:
- Do not check for latest version because there's no good way to update it when no Pi-hole docker image is available.
- Download latest `PADD` on each container start which would fix this.

yubiuser, 2023-01-20:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.1

closed 2023-01-20 by yubiuser

thomasmerz, 2023-01-20:
`Docker 2023.01.3, PADD v3.10.1`- and after a short amount of time: `Docker 2023.01.4, PADD v3.10.1` 👍🏼

PR #325 (2023-01-19): Do not show update notices if `padd` is being run from the docker container
### **What does this PR aim to accomplish?:**

Aims to resolve #324 by supressing the update available notice when PADD is run inside docker as this can cause confusion 

We can always tag a new version of the Pi-hole container to include the latest version of PADD, but it is not always possible to get to this straight away, so it is best to avoid showing users the update notice.

_(Thinking out loud: I wonder if it is worth building in logic that shows the container version/container update available instead of the re [... cut]