pi-hole/PADD #331: Screen Won't Time Out

omegaapex128, 2023-01-11:
I am using the Pi 7" touchscreen and everything is working great. I only wonder, and have been unable to find any information, if it is possible for the display to go blank after a period of say 5 minutes.  I know that there are instructions on how to make the screen go off during specific times (i.e. midnight to 8 AM) but haven't found anything else. I have tried to set the screen to go blank after 5 minutes on Raspbian itself, but it seems PADD overrides this.

Is there something simple I'm missing?

yubiuser, 2023-01-11:
This is clearly a question out -of-scope of PADD. We are not keeping the screen awake with any special command, but only updating the display every 5 second.
You would need a custom script to switch off the screen after a certain amount of time. Searching briefly revealed some issues with shutting off RPi screens depending on your OS and mode of connection.

https://github.com/raspberrypi/firmware/issues/1740

closed 2023-01-11 by yubiuser

omegaapex128, 2023-01-13:
I mean, PADD is the only program that runs that gives me this problem as I have the device to time the screen out after 5 minutes but 🤷‍♂️

yubiuser, 2023-01-13:
You need to figure out what prevents the screen from turning off: is it a screen update, is it disk activity, is it network activity? But this depends on the script/program you use to turn it off. If you know this we can have a look at PADD again.