pi-hole/PADD #333: Wiki is outdated

yubiuser, 2023-01-15:
As the title says.

mwoolweaver, 2025-02-14:
I was going update some of the links in the pages of the wiki but GitHub doesn't allow forking of wiki's.

FWIW, it seems the details of only 2 pages (known issues & sizes) in the wiki is all that's not already shown in the README.

Would it be better, or more easily maintained, if the info of those 2 pages is moved to the README?

yubiuser, 2025-02-14:
Sure. The wiki always has the tendency to go stale as nobody is really taking care of it.
If you want, you can port the relevant info into the README and we simply delete the wiki

mwoolweaver, 2025-02-14:
is this still a known issue? if not i will leave it out of the readme

------
**Issue:**
PADD will show that one or all parts of Pi-hole are out-of-date.

**Workaround**
- Exit PADD by hitting `Ctrl+C` twice.  
- Delete the piHoleVersion file with:
  `rm piHoleVersion`
- Restart PADD with:
  `./padd.sh`
------

mwoolweaver, 2025-02-14:
also worth noting is adafruit has changed the scripts in their repo to python scripts from shell scripts. 

should we include these new instructions in the readme or just link to it?

rdwebdesign, 2025-02-14:
Personally, I think a link would be fine.
There is no need to add the instructions here. This is an external code to use an optional hardware, not specific for PADD or Pi-hole.

mwoolweaver, 2025-02-14:
how does the Readme here look

https://github.com/mwoolweaver/PADD/tree/readme/consolidate-wiki

yubiuser, 2025-02-15:
> is this still a known issue?

No, it's not. I think PADD used to check Pi-hole versions by itself and wrote a file somewhere. But that file does not exist anymore:

___

> how does the Readme here look

Why did you close #418? 
Please re-open/open an other PR so we can directly make comments and suggestion to the PR

mwoolweaver, 2025-02-15:
> Please re-open/open an other PR so we can directly make comments and suggestion to the PR

i have submitted #419

yubiuser, 2025-02-24:
Wiki is deleted, all relevant information in in README