pi-hole/PADD #334: printf-error: not completely converted

marwenius, 2023-01-17:
**Describe the bug**
With each refresh these error messages are displayed:
```
./padd.sh: 184: printf: 37,552: not completely converted
./padd.sh: 914: printf: 13,0: not completely converted
./padd.sh: 941: printf: 4,5: not completely converted
./padd.sh: 941: printf: 1,3: not completely converted
```

**To Reproduce**
Steps to reproduce the behavior:

1. start ./padd.sh (no matter whether via .bashrc or manually)

**Expected behavior**
No error messages.

**Screenshots**
_none_

**Additional context**
- Raspberry Pi 3 Model B Plus Rev 1.3
- Raspberry Pi OS Lite (64-bit)
- Pi-hole v5.15
- FTL v5.20.1
- Web Interface v5.18.1
- PADD v3.10.0
- PiTFT 3.5" Resistive touchscreen

rdwebdesign, 2023-01-17:
Looks like your system is returning the values using comma as decimal separator and PAAD is not expecting these values.

What is the output of `locale` on the command line?

yubiuser, 2023-01-17:
Please change the lines at the beginning from

```
LC_ALL=C
LC_NUMERIC=C
```
to

```
export LC_ALL=C
export LC_NUMERIC=C
```

and see if this fixes your issue.

marwenius, 2023-01-17:
> Looks like your system is returning the values using comma as decimal separator and PAAD is not expecting these values.
> 
> What is the output of `locale` on the command line?

LANG=de_DE.UTF-8
LANGUAGE=
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=

> Please change the lines at the beginning from
> 
> ```
> LC_ALL=C
> LC_NUMERIC=C
> ```
> 
> to
> 
> ```
> export LC_ALL=C
> export LC_NUMERIC=C
> ```
> 
> and see if this fixes your issue.

The issue is fixed. Thank you!

yubiuser, 2023-01-20:
Fix released with https://github.com/pi-hole/PADD/releases/tag/v3.10.1

closed 2023-01-20 by yubiuser