pi-hole/PADD #339: Update to V3.10.1

coujin11, 2023-01-20:
After updating to V3.10.1 and rebooting my Pi4 - I am getting screen issues.lines

Any advise on this and is there an easy way to roll back to v3.10.0?

coujin11, 2023-01-20:
[image]

coujin11, 2023-01-20:
I have downgraded back to V3.10.0 but is problem is still present

PromoFaux, 2023-01-20:
What screen is that? Looks like it may be an issue with the hardware?

coujin11, 2023-01-20:
It was working perfectly fine before hand. I have uploaded a photo

coujin11, 2023-01-20:
Hyperpixel 4 - I was ordering completely fine before the update/reboot

yubiuser, 2023-01-20:
I agree with **PromoFauy**: this looks like a hardware issue. Esp. as the issue persists after downgrading. I think we have more users with Hyperpixel 4, let's see if they observe the same issue.

PromoFaux, 2023-01-22:
I've got one on order - will be able to confirm once it arrives

PromoFaux, 2023-01-24:
I have received it. I have no idea how to set it up. I will read the docs again later!!

PromoFaux, 2023-01-25:
I have the orientation set the opposite of yours, however all looks good here:

[image]

I suspect your screen has a hardware issue - or could be an issue with display drivers (@gadgetoid - feel free to ignore this ping if you like, but if you have any advice it would be appreciated)

You could also take a look at : https://github.com/pimoroni/hyperpixel4/issues/177

I needed to refer to that when setting up the screen on this Raspberry Pi 3b, but same applies to the 4.

nineteenletterslong, 2023-02-15:
I have a hyperpixel 4 as well. Is there a way padd can utilize the entire screen like this hack? https://gist.github.com/giuliomagnifico/91344ab9b3f304aaec5ff3fa90ad22e3

The author hasn't updated it for 3.10.1 changes and has to update every time padd updates but it would be nice if the official padd release could accommodate these changes for larger screen users

yubiuser, 2023-02-15:
Pull requests are always appreciated :⁠-⁠)

nineteenletterslong, 2023-02-23:
Took it upon myself to use a diff program to merge changes. Here's a version 3.10.1 for hyperpixel 4

https://pastebin.com/N2Nttzkt

yubiuser, 2023-02-23:
Both versions display some unbound stats without replacing them in case unbound is not installed (or `unbound-control` is not enabled). I think we should not add something to the 'official' PADD that displays infos from a certain external software. Other users might want `cloudflared` stats displayed, other XYZ stats...

nineteenletterslong, 2023-02-23:
Oh that’s right, I forgot that was an addition Giulio made, should be a quick fix

closed 2023-03-31 by github-actions[bot]