pi-hole/PADD #363: Add support for Unbound stats?

webnetwiz, 2023-08-26:
**Is your feature request related to a problem? Please describe.**
So like lots of people out there, I've been using PiHole with Unbound DNS server. I have grabbed a very elegant solution to adding Unbound stats to PADD from this source: [https://giuliomagnifico.medium.com/add-unbound-stats-to-padd-for-pi-hole-38b2ea87db2b](https://giuliomagnifico.medium.com/add-unbound-stats-to-padd-for-pi-hole-38b2ea87db2b) and everything worked like a charm I see that since the version of PADD I'm running, which is 3.6.7, a lot has changed, and it looks like simply copying and pasting the right code from the link above will not simply add Unbound stats where needed. It would be awesome to have a version of the PADD script that it tailored towards those that run PiHole with Unbound.

**Describe the solution you'd like**
Add Unbound stats to script using the following code or something better that would yield the same results:

`CleanEcho "UNBOUND ==========================================================================="
    CleanPrintf " %-10s%-19s%-10s%-9s\e[0K\\n" "Total: " "$(sudo unbound-control stats_noreset | awk -F '=' '/total.num.queries=/ {printf $NF}') queries" " Reply:" " $(sudo unbound-control stats_noreset | awk -F '=' '/total.recursion.time.avg/ {printf "%.3f\n", $NF}') seconds" 
    CleanPrintf " %-10s%-19s%-10s%-9s\e[0K\\n" "Missed: " "$(sudo unbound-control stats_noreset | awk -F '=' '/total.num.cachemiss=/ {printf $NF}') queries" " Cached:" " $(sudo unbound-control stats_noreset | awk -F '=' '/total.num.cachehits/ {printf $NF}') queries"`

This is located in the PrintNetworkInformation function in my version of the script.

**Describe alternatives you've considered**
I haven't looked at any alternatives as I haven't found anyhting better than PADD.

**Additional context**
Attaching complete version of my customized padd script.
[padd-unb.txt](https://github.com/pi-hole/PADD/files/12444175/padd-unb.txt)

THANK YOU IN ADVANCE!!! This is an amazing piece of code that does help a lot in running PiHole effectively!

yubiuser, 2023-08-26:
This request comes up from time to time but I see little chance of implementation. The overhead for implementing this is to big: we need to check if users run `unbound` at all and if they have enabled the stats. Additionally, this would require to (re-)add `sudo` requirements to `padd` which I'd like to avoid.

closed 2023-10-01 by github-actions[bot]

webnetwiz, 2023-11-08:
There seems to be a lot of guides out there pointing to this type if setup (Pihole + Unbound) when users do not want to rely on external DNS servers and want to run local DNS servers. Almost every guide I saw talks about enabling stats for Unbound. This is the reason I asked this, as I use the external DNS as forwarders, but run Unbound locally behind my Pihole (on the same RP4 box). Either way, let me know if there's a possibility that you may pick this up at some point. Thank you!

MatthK, 2024-12-11:
One up on this. Ever since I started using Pihole, I also chose the provided guide to install and configure Unbound with it. It would be really nice to see that integrated into the script.