pi-hole/PADD #368: CPU temp not updating - RPi 4

shadowa4, 2023-09-21:
I recently did a fresh install of Raspberry Pi OS on a RPi 4b in order to install the 64bit version. While following the normal padd install guide, I reached the usual padd output screen on the CLI, with all fields updating. However, I also have an Adafruit screen mounted to the Pi displaying the PADD output 24/7.

It's in the external screen that only the CPU temp fails to update. It will display correctly at boot, and remain with the same value while all other stats update as usual.

Has anyone seen this? The screen displays all other CLI outputs just fine, and even the desktop when in desktop mode. This leads me to believe its a PADD config problem, but would like some input.

Thanks.

EdmundGerber, 2023-09-22:
I'm just noticing that as well, on an older RPi2

yubiuser, 2023-09-23:
I'm not sure if this is a PADD issue. It reads the temperature correctly at boot, all subsequent reads use the same source. 

We use the following sources

https://github.com/pi-hole/PADD/blob/81aeac95d2750998e8a70e8c38cda90293cb1ff1/padd.sh#L163-L175

Please check which one exists on your system and if the OS is updating the temperature there. Also: does the Pi-hole web GUI updated the temperature?

EdmundGerber, 2023-09-23:
The Web GUI works as usual. I'm not sure when this stopped working, but it hasn't changed since the last Pihole restart. If I restart PADD, it shows the current temp, but then never changes until the next reboot.

I'm not sure how to determine what sensors this is using, but certainly the same ones they always have been. I'll see if anything needs updating, and report any change. On Raspbian Bullseye (5.15.84-v7+)

rdwebdesign, 2023-09-23:
I think I know what is the issue.

Raspberry PI use this code to update the temperature:
https://github.com/pi-hole/PADD/blob/81aeac95d2750998e8a70e8c38cda90293cb1ff1/padd.sh#L167-L169

On the first iteration, the variable starts empty and the correct value is set.

On the subsequent iterations the variable is already set and the test `[-z "${cpu}"]` fails. As a result, the value is not updated and the old value is always displayed.

This is probably happening on almost every device, except those using the sensor tested on line 164.

To fix this, we need to unset the variable before every test.

EdmundGerber, 2023-09-23:
Sounds like you are on the right path. As I had time to kill, I grabbed PADD 3.10.1, and it works as expected.

rdwebdesign, 2023-09-24:
PADD 3.10.1 temperature tests were changed because there were cases where some temp sensor files existed, but they were empty, causing errors.

rdwebdesign, 2023-09-24:
@EdmundGerber 

Can you test this version? 
- https://raw.githubusercontent.com/pi-hole/PADD/f87333191b7d3b7378a4741ea0a3c3d6a16c2b09/padd.sh

It is fixed?

EdmundGerber, 2023-09-24:
So far so good - temperature changing as it should, with the version linked. Thank you for looking into this.

yubiuser, 2023-09-24:
Good catch @rdwebdesign.

I think we could have a bug fix point release today...

PR #370 (2023-09-24): Use `unset` to reset CPU temp variable on each iteration
### What does this PR aim to accomplish?

[Fix #368](https://github.com/pi-hole/PADD/issues/368)

### How does this PR accomplish the above?

Resetting the `$cpu` variable between calls.  

### Link documentation PRs if any are needed to support this PR:

none

---
**By submitting this pull request, I confirm the following:**

1. I have read and understood the [contributors guide](https://docs.pi-hole.net/guides/github/contributing/), as well as this entire template. I understand which branch to base my commits and Pull Requests against. [... cut]
closed 2023-09-24 by yubiuser