pi-hole/PADD #380: Flickering on Inland 3.5" TFT

omegaapex128, 2024-01-12:
Saw a lot of previous issues that were reported as the same problem in which the screen flickers anywhere between 2-10 seconds as the screen seemingly is refreshing.  I have tried all the fixes in those reported issues, but being that they are all 3-4 years old, all those fixes have been implemented in the padd.sh code but it is still occurring.  PADD launches on it's own, no desktop environment.  Everything I've looked through from resolutions from 2020 and 2022, like I said, have already been put in the code so I really don't know what else I can change to stop this from happening. Not sure if it's my screen or if it's just still an issue.

yubiuser, 2024-01-12:
I'm not using an external TFT to run PADD so I have no way to investigate this myself. We've greatly improved the code which should avoid most flickering. Maybe you can record a video showing the flickering, maybe this could give us some hints

omegaapex128, 2024-01-12:
I uploaded what it's doing to my YouTube. Here's the unlisted link. https://www.youtube.com/watch?v=xgIsHCw_ScY

yubiuser, 2024-01-12:
Ok, this not PADD flickering itself, but switching back to some other console terminal (you see the boot log in the background). This is an issue outside of PADD. What OS are you running?

yubiuser, 2024-01-12:
Which command did you use to start PADD?

omegaapex128, 2024-01-12:
Raspberry Pi 3 Model B running Raspbian 11 Bullseye. I did the /.bashrc to autorun PADD

yubiuser, 2024-01-12:
What happens if you remove it from there and manually start `padd.sh` does it stay in foreground?

omegaapex128, 2024-01-12:
Well I rebooted and when I run it I see it in the Putty window on my laptop, but my display on the Pi is still on the boot log. It's no longer flickering PADD which makes sense since I removed it, but running the program in my Putty is not making it appear on the Pi display

yubiuser, 2024-01-12:
Have a look here: https://superuser.com/a/1557262
This should allow you to connect via Putty and run padd but display it on the external screen.

omegaapex128, 2024-01-12:
Do I put that in the bashrc?

yubiuser, 2024-01-12:
Not now. Try within Putty first and check if this is enough to display PADD on your external screen without flickering

omegaapex128, 2024-01-12:
Ok I tried it out and it threw the following
[image]

omegaapex128, 2024-01-12:
I went back and looked at settings and saw I had the Pi auto-logging in to console so I tried switching it to auto-logging in to desktop. PADD doesn't launch and when I reboot the Pi, I can briefly see it as if it is running behind the desktop environment before it shuts down

rdwebdesign, 2024-01-12:
I currently don't have a screen to test and I'm not sure if this will help you, but what is the output of `cat /boot/cmdline.txt`?

Do you see `console=tty0` or `console=tty1`?

omegaapex128, 2024-01-12:
console=tty1

rdwebdesign, 2024-01-12:
Hmmm... Sorry, never mind. This is the expected value. My guess was wrong.

closed 2024-02-18 by github-actions[bot]