pi-hole/PADD #414: PADD does not properly reject failed authentication when 2FA is enabled on WebUI

mwoolweaver, 2025-02-13:
**Describe the bug**
when 2FA is enabled on WebUI and you provide the WebUI password (w/o TOTP)  PADD will say `Authentication successful.` and will produce a broken dashboard as seen in screenshot.

_It appears that PADD assumes it is dealing with pi-hole in a docker container given the output shown._

**To Reproduce**
Steps to reproduce the behavior:

1. setup 2FA on WebUI 
2. start PADD (`./padd.sh --server pi.hole`)and provide WebUI password (w/o TOTP) 
3. get what is seen in screenshot

**Expected behavior**

give failed login error instead of broken output

**Screenshots**
[image]

**Additional context**

also side question:

how does PADD expect to receive the `--secret`? 

i have tried with 

double quotes `"password"`
single quotes `'password'`
no quotes `password`

it always rejects it and asks for it again then i can paste into terminal it will accept it.

yubiuser, 2025-02-13:
> also side question:

This is a bug introduced by https://github.com/pi-hole/PADD/pull/392.

Please try branch `fix/secret`

mwoolweaver, 2025-02-13:
> Please try branch `fix/secret`

that works for passing the secret

mwoolweaver, 2025-02-13:
Also just for clarity on this 

> when 2FA is enabled on WebUI and you provide the WebUI password (w/o TOTP). 

PADD rejects other wrong passwords correctly but when you provide the WebUI specifically

yubiuser, 2025-02-13:
I'll have a look at supporting 2FA/correctly rejecting 2FA.

Did you try to login with enabled 2FA and the app password?

yubiuser, 2025-02-13:
@mwoolweaver 

Could you please try `2fa` brach as well?

mwoolweaver, 2025-02-13:
2FA branch seems to require 2FA regardless of the password I provide. 

Not sure what is expected

yubiuser, 2025-02-14:
Yes, this is kind of by design. As we don't know what type of password the user inputs we ask for the second factor. However, if you supply the app password, the 2fa can be empty or some random number.

mwoolweaver, 2025-02-14:
> However, if you supply the app password, the 2fa can be empty or some random number.

can confirm passing a `0` works as you describe

mwoolweaver, 2025-02-14:
considered fixed by #416

rdwebdesign, 2025-02-18:
Probably similar to this issue: https://github.com/pi-hole/pi-hole/issues/5771

Try to put the password in single quotes, like this: `--secret '+pass/word='`

mwoolweaver, 2025-02-18:
@rdwebdesign it is fixed with #416 or #415 but from the looks of not many have encountered it yet

yubiuser, 2025-02-24:
PAdd v4 is released.

closed 2025-02-24 by yubiuser