pi-hole/PADD #421: Missing setupVars.conf breaks padd.sh after PiHole v6.0

albertzsigovits, 2025-02-18:
**Describe the bug**
After upgrading PiHole to v6.0, I wanted to use PADD.sh, just as on previous versions.
The bash script defaults to an error message:
https://github.com/pi-hole/PADD/blob/master/padd.sh#L1507

After the v6.0 upgrade, the setupVars.conf file is gone, and only a .update.bak backup file is left behind. (I don't see this being mentioned in the changelog for v6.0).
Probably need to handle the changes in v6.0 to be suitable for PADD.sh to run properly.

A workaround is to re-create the setupVvars.conf file from the .update.bak file (although not sure if it breaks anything), or just modify the script to point to a different filename.

**To Reproduce**
Steps to reproduce the behavior:

1. Upgrade PiHole to v6.0
2. Launch latest padd.sh

**Expected behavior**
Execution of padd.sh does not result in an error due to the missing setupVars.conf file

mwoolweaver, 2025-02-18:
you will need to use the development branch 

https://github.com/pi-hole/PADD/tree/development

albertzsigovits, 2025-02-18:
Awesome, thanks @mwoolweaver !

accordxtc, 2025-02-19:
Any idea on ETA, my padd currently is asking for a password, not sure what password is needed

mwoolweaver, 2025-02-19:
in most cases it will be the WebUI password you use when loggin in @ http://pi.hole

accordxtc, 2025-02-19:
Silly question for the password and I just typing 

./padd.sh --secret <password> and replace the password with mine

mwoolweaver, 2025-02-19:
./padd.sh --secret <password-here>

mwoolweaver, 2025-02-19:
are you running on the Pi-hole directly or another device?

mwoolweaver, 2025-02-19:
and try using the version found here 

https://github.com/pi-hole/PADD/blob/2fa/padd.sh 

has a fix for passing the password with `--secret`

accordxtc, 2025-02-19:
This is the version I was using. I am running this directly on its own dedicated Rpi

mwoolweaver, 2025-02-19:
as seen in the readme of the development branch

> Pi-hole v6 uses a completely new API with a new authentication mechanism
> 
> If you run PADD on the same machine as Pi-hole, it's possible to bypass authentication when your local user is member of the pihole group (specifically, if you can access `/etc/pihole/cli_password). For details see https://github.com/pi-hole/FTL/pull/1999
> 
> If this is not the case, PADD will ask you for your password and (if configured) your two factor authentication token. You can also pass those as arguments

yubiuser, 2025-02-19:
> pi@pihole64:~$ ./padd.sh
./padd.sh: line 8: syntax error near unexpected token `newline'
./padd.sh: line 8: `<!DOCTYPE html>'

You've downloaded the HTML file, not the raw file.

Andymc1280, 2025-02-19:
> you will need to use the development branch
> 
> https://github.com/pi-hole/PADD/tree/development

how do you switch to development branch (remind me) on a running install (RPi) ?

mwoolweaver, 2025-02-19:
switching branches for PADD will just require you to download the script again from the development branch

https://raw.githubusercontent.com/pi-hole/PADD/refs/heads/development/padd.sh

accordxtc, 2025-02-21:
@mwoolweaver I was able to get PADD running by adding my Rpi login to the PiHole user group. Last issue I'm having and not sure if you know. It was easy in previous versions to get / show API token. Do you know if this is still possible. I was using this to have Home Assistant showing PiHole stats

yubiuser, 2025-02-21:
Home assistant integration likely need some adoption to the new authentication system. See here for an example: https://docs.pi-hole.net/guides/misc/homeassistant/

yubiuser, 2025-02-24:
PADD v4 is released

closed 2025-02-24 by yubiuser