pi-hole/PADD #425: push development branch to master

robcmo, 2025-02-24:
Now that Pi-Hole 6 is released, the release version of padd.sh does not work.  Installing the dev branch does work, except the loss of temperature which was said to be deprecated so lines 416-417 need to be removed.

mwoolweaver, 2025-02-24:
[image]

as seen here https://github.com/pi-hole/web/issues/3218#issuecomment-2667112651

> Removal of the temp icon and value on the web interface is by design. It's still present in the API - we left it exclusively for PADD.

robcmo, 2025-02-24:
> > Removal of the temp icon and value on the web interface is by design. It's still present in the API - we left it exclusively for PADD.

@mwoolweaver, that's great news, but it no longer works for me. I will check the logs.

[image]

yubiuser, 2025-02-24:
https://github.com/pi-hole/PADD/pull/428 will fix this

XhmikosR, 2025-03-04:
@yubiuser this can be closed now :)

closed 2025-03-04 by yubiuser