pi-hole/PADD #431: Provide option to mimic discontinued chronometerand exit PADD when run

daPhipz, 2025-02-24:
I used the old chronometer as part of my MOTD when SSHing into the Pi running Pi-Hole. This gave me a quick overview about the current status and load on the Pi.

I have never heard of PADD before, and now with Pi-Hole 6 out, the old chronometer is discontinued. When installing and running PADD (v4), it seems to be running in a loop and always updating the displayed info regularly. Pressing Ctrl-C gives me my prompt back.

Is there a way to have PADD run only once, exit the program and show me my prompt again? That way, I can use it as a one-time quick look whenever I SSH into the Pi running Pi-Hole and then use my Pi in another way - as I was doing before.

yubiuser, 2025-02-24:
Currently there is no 'run-once-and-exit' option which prints everything nicely. You could use the `-j` option for some basic stats

```
./padd.sh -j
Establishing connection with FTL...
No authentication required.
{"domains_being_blocked":1003065,"dns_queries_today":32935,"ads_blocked_today":3998,"ads_percentage_today":12.1,"clients": 10}
```

k-rister, 2025-02-27:
I hacked this together really quickly for similar reasons:

```
diff --git a/padd.sh b/padd.sh
index 8fa5ef8..b7f1307 100755
--- a/padd.sh
+++ b/padd.sh
@@ -17,6 +17,8 @@ export LC_NUMERIC=C
 # VERSION
 padd_version="v4.0.0"

+runOnce=0
+
 # LastChecks
 LastCheckPADDInformation=$(date +%s)
 LastCheckFullInformation=$(date +%s)
@@ -1692,6 +1694,10 @@ NormalPADD() {
     # Output everything to the screen
     PrintDashboard ${padd_size}

+    if [ "${runOnce}" -eq 1 ]; then
+       break
+    fi
+
     # Sleep for 5 seconds
     # sending sleep in the background and wait for it
     # this way the TerminalResize trap can kill the sleep
@@ -1802,6 +1808,7 @@ DisplayHelp() {
 :::  --server <DOMAIN|IP>    domain or IP of your Pi-hole (default: localhost)
 :::  --secret <password>     your Pi-hole's password, required to access the API
 :::  --2fa <2fa>             your Pi-hole's 2FA code, if 2FA is enabled
+;;;  --runonce               display output once and exit
 :::  -j, --json              output stats as JSON formatted string and exit
 :::  -u, --update            update to the latest version
 :::  -v, --version           show PADD version info
@@ -1898,6 +1905,7 @@ while [ "$#" -gt 0 ]; do
     "-u" | "--update"   ) Update;;
     "-h" | "--help"     ) DisplayHelp; exit 0;;
     "-v" | "--version"  ) xOffset=0; ShowVersion; exit 0;;
+    "--runonce"         ) runOnce=1;;
     "--xoff"            ) xOffset="$2"; xOffOrig="$2"; shift;;
     "--yoff"            ) yOffset="$2"; yOffOrig="$2"; shift;;
     "--server"          ) SERVER="$2"; shift;;

```

yubiuser, 2025-02-27:
Would you like to file a pull request?

k-rister, 2025-02-27:
Sure.  When I have some more time I was going to make sure I followed the coding style, etc. and then do so.

robcmo, 2025-03-01:
@daPhipz, the timeout command is a good solution for you. PADD takes around 10 seconds to startup.

`timeout 12 ./padd.sh`

daPhipz, 2025-03-05:
> `timeout 12 ./padd.sh`

Thanks @robcmo, this is a good workaround!

daPhipz, 2025-04-05:
As I understand it, this issue will get a fix with #443, which will land in version 4.1.0 as per #438 - correct?

yubiuser, 2025-04-05:
Correct.

daPhipz, 2025-05-06:
Still open, waiting for the new release PR to be merged.

closed 2025-06-12 by github-actions[bot]

yubiuser, 2025-06-16:
I'll try to give PADD some attention in the next days including a new release which will fix this issue.