pi-hole/PADD #441: Update screenshot for v4.x

XhmikosR, 2025-03-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mwoolweaver, 2025-11-07:
my attempt can be seen here

https://github.com/mwoolweaver/graphics/commit/cafd6ba5c8bb34e4add527cc2da02c8b494aa745

I'm sure someone could potentially do better.

mwoolweaver, 2025-11-07:
my attempt can be seen here. 

commit for image compare
https://github.com/mwoolweaver/graphics/commit/3abe8c741a3b1226ee034038bc59ca8b6e085017#diff-6465b027f656c00d7f1fcb9bb8131d568d86dda23d36638fa3e33b389c566426

on a readme for compare as well
https://github.com/mwoolweaver/PADD/tree/image-test

rdwebdesign, 2025-11-07:
I don't know why, but your images seems a little blurry.