pi-hole/PADD #444: Padd starts on boot but gives error then fails

LuxStatera, 2025-03-06:
**Describe the bug**
A clear and concise description of what the bug is.

I have Padd running on boot with bash.rc and it definitely starts Padd on boot. Unfortunately there seems to be an authorization error. It won't connect and then ends Padd. I have a feeling it thinks I'm running Padd and Pihole on different devices.

**To Reproduce**
Steps to reproduce the behavior:

1. I added "bash /home/pi/padd.sh --secret *******" to bash.rc to get Padd to run on boot
2. sudo reboot

**Expected behavior**
A clear and concise description of what you expected to happen.

I expected Padd to start up and auto login with my provided password

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Additional context**
Add any other context about the problem here.

Once the Pi reboots, Padd starts up but it gives me an error and then fails. It gives this error:
curl: (3) URL using bad/illegal format or missing URL
curl: (6) Could not resolve host: communicat ionsauth
curl: (6) Could not resolve host: errorauth
curl: (6) Could not resolve host: toauth
curl: (3) URL using bad/illegal format or missing URL
curl: (6) Could not resolve host: timedauth
curl: (6) Could not resolve host: outauth
Establishing connection with FTL...
curl: (6) Could not resolve host: outauth
No response from FTL server. Please check connectivity and u
se the options to set the API URL
Usage: /home/pi/padd.sh [--server <domainlIP>]

I'm running Padd on the same Pi that is running Pihole. What do I need to do to fix this?

yubiuser, 2025-03-06:
Can you run PADD manually without issues?

LuxStatera, 2025-03-06:
Yes although it requires a password each time

LuxStatera, 2025-03-06:
When I use the --secret flag it works without issue

yubiuser, 2025-03-06:
Good. 

Maybe it's a timing issue. I suspect PADD starts before FTL is up and fails to authenticate. Try adding a sleep time before you start PADD on startup

LuxStatera, 2025-03-06:
That worked! Thank you!

closed 2025-03-06 by LuxStatera