pi-hole/PADD #445: PADD.sh local fails, but remote with --server succeeds (jq on rasbian buster too old?)

cwilson1776, 2025-03-07:
**Describe the bug**
I have pi-hole running on a Raspi3b (with a PiTFT display, like [this](https://learn.adafruit.com/pi-hole-ad-pitft-tft-detection-display/raspberry-pi-setup)). Executing PADD.sh (4.0.0, pi-hole v6) locally doesn't work (see screenshots); however PADD.sh 3.x + pihole v5 worked on the same platform.

[image]
[image]

Note that I am getting one response -- the latest blocked site value is populated, but nothing else. A little 'echo'-style debugging shows that the problem appears to be in GetPADDData: the value of `response` has all the data you'd want, but the value of `padd_data` has only a single line:

`padd_data='recent_blocked=scribe.logs.roku.com'`

On this machine (running raspbian buster), /bin/sh is a symlink to dash.
- dash: version 0.5.10.2-5
- jq: version 1.5+dfsg-2+b1

What's strange is, I can use PADD.sh 4.0.0 with the --server option from a different machine, and it works.

[image]
[image]

On this remote machine (ubuntu jammy), /bin/sh is again a symlink to dash.
- dash: version 0.5.11+git20210903+057cd650a4ed-3build1
- jq: version 1.6-2.1ubuntu3

**To Reproduce**
Steps to reproduce the behavior:

1. Install raspbian buster
2. echo '{"recent_blocked":"scribe.logs.roku.com","active_clients":56,"gravity_size":1546410}' |\
jq -r 'paths(scalars | true) as $p | [$p | join(".")] + [if getpath($p)!=null then getpath($p) else "null" end] | join("=")' 2>/dev/null
3. see only 'recent_blocked=scribe.logs.roku.com'

**Expected behavior**
3. See 'recent_blocked=scribe.logs.roku.com
active_clients=56
gravity_size=1546410'

I think the issue is that jq must be (some version above 1.5?). I don't really think there is a fix for this, other than "try compiling a newer version of jq since buster doesn't provide one'

I'll give that a shot and report back.

cwilson1776, 2025-03-07:
`sudo wget -O /usr/local/bin/jq https://github.com/jqlang/jq/releases/download/jq-1.7.1/jq-linux-armhf`

... and now it works.

yubiuser, 2025-03-07:
Thanks for confirming.

closed 2025-03-07 by yubiuser