pi-hole/PADD #446: PADD v4.0 Slim truncates top line on 24x80 screen

bob787, 2025-03-07:
Happy PADD user since 2017. Since Pihole v6/PADD v4, my 3.2" Adafruit screen (stty size yields 24x80) cuts off the top line in favor of a blank line below a fully resolved IPv6 address (see photo)

[image]

Can anything be done to remove the IPv6 line (I do not use IPv6) or the blank line below it to get the screen back to normal?

Let me know if you need any more information,

yubiuser, 2025-03-07:
Mhhh... 80x24 should be regular, not slim. But looking at your screenshot, the order of items suggest it's actually 'tiny'

rdwebdesign, 2025-03-07:
The screen on the image is showing 52x20.

bob787, 2025-03-07:
Right you are... I counted the characters and the screen is displaying 52x20 (despite `stty size` reporting `24 80`. 

I also tried `sudo dpkg-reconfigure console-setup` with UTF-8, Guess, Terminus, 6x12 with no change

Can anything be done to get the display back to normal? Thanks for looking

bob787, 2025-03-07:
Reading your comments inspired me to start reading through the code and while I did not understand most of it, I was able to blow away the two lines from the 'tiny' section that were creating the issue and got a clean screen again. Thanks for helping and have a great weekend!

[image]

closed 2025-03-07 by bob787