pi-hole/PADD #448: —server doesnt work when pihole is running locally

darkexplosiveqwx, 2025-03-17:
**Describe the bug**
running `./padd.sh --server 192.168.178.88 --secret shh` displays information on the pihole running locally
**To Reproduce**
Steps to reproduce the behavior:
I have pihole running locally and my user is part of the `pihole` group.
When running `./padd.sh --server 192.168.178.88 --secret  shh` i get information displayed about the local pihole not the one on .88
The ip needs to be running pihole, for example 
`./padd.sh --server 192.168.178.89 --secret shh` will result padd on getting stuck on the loading screen.

**Expected behavior**
Display information about 192.168.178.88

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

yubiuser, 2025-03-17:
Please run 

```
bash -x ./padd.sh --server 192.168.178.88 --secret shh
```

And post the last lines of output

darkexplosiveqwx, 2025-03-17:
I think I know what the issue is: 
```

+ echo 'START UP ==================================================='
+ TestAPIAvailability
+ local chaos_api_list authResponse cmdResult digReturnCode authStatus authData
+ '[' -z 192.168.178.88 ']'
+ '[' 192.168.178.88 = localhost ']'
+ '[' 192.168.178.88 = 127.0.0.1 ']'
++ dig +short chaos txt domain.api.ftl @192.168.178.88
++ echo 0
+ cmdResult='"https://pi.hole:443/api/"
0'
++ echo '"https://pi.hole:443/api/"
0'
++ tail -n 1
+ digReturnCode=0
+ '[' '!' 0 = 0 ']'
++ echo '"https://pi.hole:443/api/"
0'
++ head -n 1
+ chaos_api_list='"https://pi.hole:443/api/"'
+ '[' -n '"https://pi.hole:443/api/"' ']'
+ API_URL='"https://pi.hole:443/api/"'
+ API_URL='"https://pi.hole:443/api/'
+ API_URL=https://pi.hole:443/api/
++ curl --connect-timeout 2 -skS -w '%{http_code}' https://pi.hole:443/api/auth
```
The dig command got answered with pi.hole, since but this machine is using itself and 1.1.1.1 as dns, pi.hole will resolve to be itself.

More of the output if needed:
```
+ '[' 4 -gt 0 ']'
+ case "$1" in
+ SERVER=192.168.178.88
+ shift
+ shift
+ '[' 2 -gt 0 ']'
+ case "$1" in
+ password='shh dont tell'
+ shift
+ shift
+ '[' 0 -gt 0 ']'
+ main
+ check_dependencies
+ command -v curl
+ command -v jq
+ printf '\e[?25l'
+ trap CleanExit EXIT
+ trap sig_cleanup INT QUIT TERM
++ stty -g
+ stty_orig=4500:5:bf:8a3b:3:1c:7f:15:4:0:1:0:11:13:1a:0:12:f:17:16:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0
+ SizeChecker
+ sleep 0.1
++ tput cols
+ console_width=111
++ tput lines
+ console_height=43
+ '[' 111 -ge 80 ']'
+ '[' 43 -ge 26 ']'
+ padd_size=mega
+ width=80
+ height=26
+ xOffset=15
+ yOffset=8
+ '[' -n '' ']'
+ '[' -n '' ']'
+ StartupRoutine mega
+ '[' mega = ants ']'
+ printf '\e[H\e[2J\e[3J'
[... 186 lines cut ...]
		"csrf":	"BATA+QWaOo9tZFwMUFudiA=",
		"validity":	86400,
		"message":	"password correct"
	},
	"took":	3.5524368286132812e-05
}'
++ jq --raw-output .error.message
+ sessionError=null
+ '[' '!' true = true ']'
+ moveXOffset
+ '[' 15 -gt 0 ']'
+ printf '\e[%sC' 15
+ echo 'Authentication successful.'
+ moveXOffset
+ '[' 15 -gt 0 ']'
+ printf '\e[%sC' 15
+ echo '- Requesting PADD information...'
+ GetPADDData
+ local response
++ GetFTLData padd ''
++ local response
++ local data
++ local status
+++ curl --connect-timeout 2 -sk -w '%{http_code}' -X GET https://pi.hole:443/api/padd -H 'Accept: application/json' -H 'sid: 9aM+/b83BhZv/J3Q9ft9FA=' 
```

yubiuser, 2025-03-17:
So you're having two Pi-holes in your network with both using the domain `pi.hole`?
Does it work when you change the `webserver.domain` on `.88` to something else?

darkexplosiveqwx, 2025-03-17:
I changed webserver.domain to eap.lan, this fails because eap.lan has no dns entry on localhost

Potential fix? Trying to resolve the domain via the server set with —server instead of implying that the host machine uses the pihole its PADDing to as DNS?

```
+ echo 'START UP ============================================'
+ TestAPIAvailability
+ local chaos_api_list authResponse cmdResult digReturnCode authStatus authData
+ '[' -z 192.168.178.88 ']'
+ '[' 192.168.178.88 = localhost ']'
+ '[' 192.168.178.88 = 127.0.0.1 ']'
++ dig +short chaos txt domain.api.ftl @192.168.178.88
++ echo 0
+ cmdResult='"https://eap.lan:443/api/"
0'
++ echo '"https://eap.lan:443/api/"
0'
++ tail -n 1
+ digReturnCode=0
+ '[' '!' 0 = 0 ']'
++ echo '"https://eap.lan:443/api/"
0'
++ head -n 1
+ chaos_api_list='"https://eap.lan:443/api/"'
+ '[' -n '"https://eap.lan:443/api/"' ']'
+ API_URL='"https://eap.lan:443/api/"'
+ API_URL='"https://eap.lan:443/api/'
+ API_URL=https://eap.lan:443/api/
++ curl --connect-timeout 2 -skS -w '%{http_code}' https://eap.lan:443/api/auth
curl: (6) Could not resolve host: eap.lan
+ authResponse=000
++ printf %s 000
++ tail -c 3
+ authStatus=000
++ printf %s ''
+ authData=
+ '[' '!' 000 = 200 ']'
+ '[' '!' 000 = 401 ']'
+ API_PORT=
+ local last_api_list
+ last_api_list='"https://eap.lan:443/api/"'
+ chaos_api_list='"https://eap.lan:443/api/"'
+ '[' '"https://eap.lan:443/api/"' = '"https://eap.lan:443/api/"' ']'
+ chaos_api_list=
+ '[' -n '' ']'
+ '[' -n '' ']'
+ moveXOffset
+ '[' 47 -gt 0 ']'
+ printf '\e[%sC' 47
+ printf %b 'Establishing connection with FTL...\n'
+ LoginAPI
+ '[' '' = false ']'
+ '[' -r /etc/pihole/cli_pw ']'
++ cat /etc/pihole/cli_pw
+ password=lcncqD/F6Em+7RruSYUGzEaR3ogsyDCo5CWs6j84tRo=
+ needTOTP=false
+ '[' -z lcncqD/F6Em+7RruSYUGzEaR3ogsyDCo5CWs6j84tRo= ']'
+ '[' false = true ']'
+ Authenticate
++ curl --connect-timeout 2 -skS -X POST https://eap.lan:443/api/auth --user-agent 'PADD v4.0.0' --data '{"password":"lcncqD/F6Em+7RruSYUGzEaR3ogsyDCo5CWs6j84tRo=", "totp":null}'
curl: (6) Could not resolve host: eap.lan
+ sessionResponse=
+ '[' -z '' ']'
+ moveXOffset
+ '[' 47 -gt 0 ']'
+ printf '\e[%sC' 47
+ echo 'No response from FTL server. Please check connectivity and use the options to set the API URL'
+ moveXOffset
+ '[' 47 -gt 0 ']'
+ printf '\e[%sC' 47
+ echo 'Usage: ./padd.sh [--server <domain|IP>]'
+ exit 1
+ CleanExit
+ err=1
+ trap '' EXIT INT QUIT TERM
++ stty -g
+ '[' 4500:5:bf:8a3b:3:1c:7f:15:4:0:1:0:11:13:1a:0:12:f:17:16:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0 '!=' 4500:5:bf:8a3b:3:1c:7f:15:4:0:1:0:11:13:1a:0:12:f:17:16:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0:0 ']'
+ printf '\e[?25h'
+ kill '{}'
+ DeleteSession
+ '[' '' = true ']'
+ echo
+ exit 1
```

yubiuser, 2025-03-18:
We are likely not going to change the response for `domain.api.ftl`. So we need to adjust PADD, or you create a host entry for the second Pi-hole domain on the first device.

darkexplosiveqwx, 2025-03-18:
I submitted #450 , which removes the dns lookup for `webserver.domain`

yubiuser, 2025-03-19:
Thank you. I'll review after my vacation in around 2 weeks

PR #450 (2025-03-29): Change Domain to IP in `API_URL`
## Thank you for your contribution to the Pi-hole Community!

Please read the comments below to help us consider your Pull Request.

We are all volunteers and completing the process outlined will help us review your commits quicker.

**Please make sure you**

 1. Base your code and PRs against the repositories developmental branch.
 2. [Sign Off](https://docs.pi-hole.net/guides/github/how-to-signoff/) all commits as we enforce the [DCO](https://docs.pi-hole.net/guides/github/dco/) for all contributions
 3. [Sign](https://docs.github.com/en/authentication/managi [... cut]
closed 2025-03-29 by darkexplosiveqwx