pi-hole/PADD #449: PADD v4.0.0 doesn't cleanup session when invoked on Pihole V6

thomasmerz, 2025-03-18:
**Describe the bug**
I'm collecting some stats by this cronjob and saw by accident that PADD v4 doesn't end its session
> echo $(date +\%Y-\%m-\%d) $(docker exec --tty pihole padd -j|ack domains_being_blocked) >> ~/logs/pihole-padd.log

**To Reproduce**
Run `echo $(date +\%Y-\%m-\%d) $(docker exec --tty pihole padd -j|ack domains_being_blocked) >> ~/logs/pihole-padd.log` daily and after each run you can still see the PADD session under "https://yourpi.example.com/admin/settings/api"

**Expected behavior**
Sessions should be closed after each (successfull) run.

**Screenshots**

[image]

**Additional context**
Without cleaning up session manually I'm unable to login to GUI on another device than my first session has.

yubiuser, 2025-03-18:
This is indeed a bug/oversight. The exit traps are not defined in the json function.
I don't think this will be fixed given https://github.com/pi-hole/PADD/pull/434

You should not use PADD to query the API.
Use `pihole api xyz` instead.

thomasmerz, 2025-03-18:
I'm not querying the API, I'm "displaying stats about my Pi-hole" 😉 which is exactly the purpose of `PADD`:

```
pihole-merz-nimbus:/# padd -h

::: PADD displays stats about your Pi-hole!
:::
:::
::: Options:
:::  --xoff [num]    set the x-offset, reference is the upper left corner, disables auto-centering
:::  --yoff [num]    set the y-offset, reference is the upper left corner, disables auto-centering
:::
:::  --server <DOMAIN|IP>    domain or IP of your Pi-hole (default: localhost)
:::  --secret <password>     your Pi-hole's password, required to access the API
:::  --2fa <2fa>             your Pi-hole's 2FA code, if 2FA is enabled
:::  -j, --json              output stats as JSON formatted string and exit
:::  -u, --update            update to the latest version
:::  -v, --version           show PADD version info
:::  -h, --help              display this help text
```

I let it also run as a systemd-service on TTY11 on my host-computer:
```
$ systemctl status *padd*
● pihole-padd.service - PADD - A more advanced version of the chronometer provided with Pihole.
     Loaded: loaded (/etc/systemd/system/pihole-padd.service; enabled; preset: disabled)
     Active: active (running) since Mon 2025-03-17 07:15:07 CET; 1 day 4h ago
   Main PID: 24744 (sh)
      Tasks: 11 (limit: 4915)
        CPU: 12.706s
     CGroup: /system.slice/pihole-padd.service
             ├─24744 /usr/bin/sh -c "/usr/bin/docker exec --tty pihole padd --xoff 0 --yoff 0 > /dev/tty11 < /dev/tty11"
             └─24746 /usr/bin/docker exec --tty pihole padd --xoff 0 --yoff 0

Mar 17 07:15:07 merz-nimbus systemd[1]: pihole-padd.service: Scheduled restart job, restart counter is at 2.
Mar 17 07:15:07 merz-nimbus systemd[1]: Started PADD - A more advanced version of the chronometer provided with Pihole..
```

And I do on a regular base collect and persist some stats via:
```
$ docker exec -it pihole padd -j
Establishing connection with FTL...
Authentication successful.
{"domains_being_blocked":11208216,"dns_queries_today":133156,"ads_blocked_today":15627,"ads_percentage_today":11.7,"clients": 21}

Session successfully deleted.
```

So sometimes is seems to successfully delete Session and sometimes not!? 🤷‍♂

thomasmerz, 2025-03-18:
When dropping JSON output it would really be great if you could provide samples of how to achieve this removed feature via Pihole's API.

yubiuser, 2025-03-18:
> So sometimes is seems to successfully delete Session and sometimes not!? 

Maybe it's the pipe you use to extract certain information.

___

> would really be great if you could provide samples of how to achieve this removed feature via Pihole's API.

As I said above `pihole api xyz` e.g.

```
pihole api stats/summary
```

closed 2025-03-18 by thomasmerz

thomasmerz, 2025-04-30:
@yubiuser , when using `padd` as a systemd-service on TTY11 for example at home or on a cloudserver it also leaves an active session left for each restart/reboot. This should also concern all Raspi-Users with an external display for `padd` 🤔 

[image]

```
# cat /etc/systemd/system/pihole-padd.service
# PADD - A more advanced version of the chronometer provided with Pihole.

[Unit]
Description=PADD - A more advanced version of the chronometer provided with Pihole.
#Requires=pihole.service
After=docker.service network-online.target dhcpd.service pihole.service

[Service]
Type=simple
Environment=TERM=linux
User=root
TTYPath=/dev/tty11
ExecStart=/usr/bin/sh -c "/usr/bin/docker exec --tty pihole padd --xoff 0 --yoff 0 > /dev/tty11 < /dev/tty11"
ExecStop=/usr/bin/sh -c "/usr/bin/clear > /dev/tty11 < /dev/tty11"
Restart=always
RestartSec=1
TTYReset=yes
TTYVHangup=yes
TTYVTDisallocate=yes
KillSignal=SIGHUP

[Install]
WantedBy=multi-user.target
```

Is there a solution for this or is this also deprecated?

yubiuser, 2025-05-03:
I do not use PADD as a systemd-server, but looking at your service file the `ExecStop` line looks suspicious. You do not stop PADD but rather only clear/close the TTY. I think you need to modify this line to really stop PADD.

P.S. I'm also not sure what your goal is here. It seems you created a systemd-service to connect to a Pi-hole docker container to invoke the baked-in PADD and redirected it's in/output to tty11. This looks quite complicated compared to just run PADD bare-metal on the device where you want to display it and use the `--server` option to connect to Pi-hole.

thomasmerz, 2025-05-05:
@yubiuser - this `ExecStop` gave me the hint I needed so I god rid of so many inactive sessions now. Thanks!

Regarding my goal: I'm running `PADD` also on my virtual cloudservers where I don't have physical access to any connected display. But I've access via Webconsole to all local tty.