pi-hole/PADD #452: [Discussion] Should PADD adjust to API privacy levels?

mwoolweaver, 2025-03-31:
When Pi-hole Privacy Level is set to something other than `Show everything and record everything` the values for `domains` and `clients` are shown as `null` in PADD.

**To Reproduce**
Steps to reproduce the behavior:

1. Change Pi-hole Privacy Level to something other than `Show everything and record everything`
2. See the following (depending on setting chosen obviously) 

this is with `Hide domains and clients` selected.
 
[image]

**Expected behavior**
Unsure?? Open discussion here.

rdwebdesign, 2025-03-31:
I don't think we need the privacy level in `/padd` endpoint. The API is already hiding the domains if a different privacy level is selected.

We could simply replace `null` with `hidden due to privacy`, or `---`, or `N/A`, or a different string.

yubiuser, 2025-03-31:
> We could simply replace null with hidden due to privacy, or ---, or N/A, or a different string.

We use `N/A` already when FTL is down and nothing was returned for `/padd`.

Including the privacy level would help to distinguish between those two.

yubiuser, 2025-03-31:
Using https://github.com/pi-hole/FTL/pull/2402 and https://github.com/pi-hole/PADD/pull/453

[image]

yubiuser, 2025-03-31:
We truncate down to 29 chars min. This should fit

https://github.com/pi-hole/PADD/blob/600aecb4edd19cad3a1e496b275773d1d4486f84/padd.sh#L924

closed 2025-04-07 by mwoolweaver