pi-hole/PADD #455: Missing required packages in setup/install documentation

acolwill, 2025-04-02:
**Describe the bug**
Missing required packages in setup/install documentation

**To Reproduce**
Steps to reproduce the behavior:

1. Install a "bare metal" RaspberryPi without desktop or other packages.
2. Download PADD
3. Run PADD to connect and display information against a separate PiHole instance.

**Expected behavior**
PADD should execute and show information

**Additional context**
Package "jq" is noted as being missing when PADD is first run.
"dig" is then noted as being missing when PADD is next run. This is part of the dnsutils package.

mwoolweaver, 2025-04-02:
needs to be added here

https://github.com/pi-hole/PADD/blob/47bd6c68907ddaacd39a2f363f30a571162a8aea/padd.sh#L1467-L1478

mwoolweaver, 2025-04-02:
should the `exit 1` be moved out of the `if` statements and to the end of the function to check for all depends on first run?

acolwill, 2025-04-02:
Just needs the install documentation updated with the requirements for jq, dig (dnstools) and curl on a minimal system.

mwoolweaver, 2025-04-02:
```
check_dependencies() { 
     local hasDeps=true
     # Check for required dependencies 
     if ! command -v curl >/dev/null 2>&1; then 
         printf "%b" "${check_box_bad} Error!\n    'curl' is missing but required.\n" 
         hasDeps=false
     fi
  
     if ! command -v jq >/dev/null 2>&1; then 
         printf "%b" "${check_box_bad} Error!\n    'jq' is missing but required.\n" 
         hasDeps=false
     fi

     if ! command -v dig >/dev/null 2>&1; then 
         printf "%b" "${check_box_bad} Error!\n    'dig' is missing but required.\n" 
         hasDeps=false
     fi
     if [ "${hasDeps}" = false ]; then 
         printf "%b" "${check_box_bad} Error!\nPlease the dependencies noted above\n" 
         exit 1
     fi
 } 
```

could do something similar to this