pi-hole/PADD #474: Padd telemetry not showing any relevant data

mokrunka, 2025-09-24:
**Describe the bug**
A clear and concise description of what the bug is.
All key telemetry items are blank or displacing garbage data.
**To Reproduce**
Steps to reproduce the behavior:
launch padd.sh in CLI

**Expected behavior**
The PADD utility should accurately show blocked domains, number of queries blocked, temperatures, DNS status, etc.

**Screenshots**

[image]

**Additional context**

[image]

rdwebdesign, 2025-09-24:
Do you see any data on the Pi-hole web interface?

mokrunka, 2025-09-24:
> Your Pi-hole components are out of sync. I would run `pihole -up` to update it and check if this improves things.

I'll try this again in a few. Today was my biannual update check that I do, and so I did run pihole -up today to result in the versioning you see. Also, I of course updated padd as well. 

I also get that prompt that my OS is not supported requiring me to run another command to ignore version check when updating. The original install is 4+ yrs old.

mokrunka, 2025-09-25:
Message received folks, thanks for the input. I don't monitor the latest releases (shame on me) for the pi os or Debian so I had no idea I was so behind. Honestly the pihole stuff runs perfectly and I never have any issues with it so I fell into the "don't fix it if it's not broke" approach. When I get some time, I'll think about how I want to approach the upgraded OS. I have 2 redundant rpis running pihole so hopefully I can just do them one at a time with no downtime.

closed 2025-10-26 by yubiuser